Nakakiyo092/usb2canfdv1 #13: SLCAN improvements by adding device time getting command.

Maksons, 2025-07-11:
Hi,

It would be nice to have extra command to get device time only, so we can always quickly sync with host.

Nakakiyo092, 2025-07-13:
Hello,

I agree your idea.
I will add a new feature to the `Z` command so the device returns the current timestamp when used without any option (i. e., `Z[CR]`).

Regards.

Nakakiyo092, 2025-08-30:
Hello @Maksons ,

Thank you very much for your feature suggestion!
While testing another project, I found that this feature to be extremely helpful.
I put [link](https://github.com/Nakakiyo092/canable2-fw/issues/70#issuecomment-3239309449) to the test report just in case you are interested.

Regards.

Maksons, 2025-09-25:
Hello @Elmue ,
Timestamps in µs precision is not bad idea, I have project where it's really needed for both sent out and received packets.
And I agree, timestamps are not needed for all messages and it would be great if it existence could be defined with filter.
I know STM32G4 have multiple interrupt lines where could be High Priority and Normal messages processed, and some with built in timestamp even (from TIMx).
About ASCII and 16 bytes, here will be another idea to sent only meaning part of timestamp (not whole 64bits) :)
Just sequence of  "<<" and "OR" operation at each timestamp received hex character.

PR #14 (2025-07-19): Add a new feature to the Z command so the device returns the current timestamp when used without any option (i. e., Z[CR]).
close #13